OpenScan-org/OpenScan-Design #11: 15cmx15cm 3D printer support

hadess, 2023-07-27:
I went through the v2 mini STL files, and it seems that the only thing that wouldn't fit on the bed of my Flashforge Adventurer 3 (15x15x15cm max) would be the rotor.

It would be great if this part could be printed as 2 separate parts to be joined when assembling.

(or maybe a way to buy just the rotor in the 3D printed parts on the shop? :)

probably-Erwins-Cat, 2023-07-29:
I honestly didn't think of smaller bed sizes than approx. 20x20cm for the Mini (and 30x30 for a potential Midi version).
I can have a look into splitting the rotor in smaller segments that can be reassembled. Nonetheless, a segmented rotor will not have the same dimensional accuracy and sturdiness as a single-segment rotor.

probably-Erwins-Cat, 2023-08-02:
Hey hadess,
I have designed a splitted rotor that would fit a 15x15cm build plate and will be fixated via CA glue and spare filament pieces as makeshift dowels. I will upload the rotor soonish(TM).

hadess, 2023-08-02:
Beautiful! I'll have to give it a spin soon, thanks for your work!

probably-Erwins-Cat, 2023-08-05:
[image]

closed 2023-08-05 by probably-Erwins-Cat