OpenScan-org/OpenScan-Design #17: cog-Wheel to big[DESIGN]

HartmutA, 2023-10-02:
In my case the the cog-wheel for the OpenScan Mini V2 was to big

i increased the hole for the D-shaft radius 0.15 and printed it with a scale of 99%

Do you allow that your submitted design can be freely used, published and distributed as part of the OpenScan project? (If no, please specify the licence)
yes

If applicable, add screenshots to help explain. (Drag and drop)

[image]

HartmutA, 2023-10-02:
[99p_cog-wheel.stl.txt](https://github.com/OpenScan-org/OpenScan-Design/files/12781497/99p_cog-wheel.stl.txt)
Rename to .stl and print it with a scale 99%

probably-Erwins-Cat, 2023-10-04:
0.15mm is within the resolution of printing smaller holes of 3D printers. A small change in extrusion multiplier, retraction, water content of the filament, etc. and the inner hole can nominally change in size. This is, unfortunately, an issue that has to be solved by everyone individually and cannot really be solved nicely by sizing the hole larger. (This will result in other users running into too large holes and potential slipping)

closed 2023-12-28 by probably-Erwins-Cat