OpenScan-org/OpenScan-Design #18: rpi-cover-v1 does not fit Pi3 (OpenScan Mini V2)

HartmutA, 2023-10-02:
In my case I had to change some things to fit with my Pi3

I changed following things:
1. I increased the mounting holes to 3.5mm
2. I increased the holes for the mounting screw head to 7mm
3.  I increased the distance for the PCB to 3 mm
4. I increased the hole for the HDMI connector
5. I modified the opening for the USB and RJ45 connector
6. I added 1mm more hight
7. I made a space sleeve 12mm for the distance between Pi and Shield

Do you allow that your submitted design can be freely used, published and distributed as part of the OpenScan project? (If no, please specify the licence)
Yes

If applicable, add screenshots to help explain. (Drag and drop)
[image]

[rpi-cover-v1.stl.txt](https://github.com/OpenScan-org/OpenScan-Design/files/12781685
[Distance.stl.txt](https://github.com/OpenScan-org/OpenScan-Design/files/12781699/Distance.stl.txt)
/rpi-cover-v1.stl.txt)

[rpi-cover-v1.stp.txt](https://github.com/OpenScan-org/OpenScan-Design/files/12781692/rpi-cover-v1.stp.txt)

probably-Erwins-Cat, 2023-10-04:
I will look into the backward compatibility of the shields for RPI 3b+. I realized that the RPI foundation changed the layout significantly from the 3b+ to 4 and (again) partially back again with the new RPI 5.
If you are fine, I would take your STLs (or STEP, in case you have them) and and them to the V2 STLs files as provisional versions until I find the time to adopt, re-measure and re-model the parts.

HartmutA, 2023-10-05:
Hi probably Erwin, i added the files in the previus post. I had to add a ".txt" and it looks a little bit mixed. but the links are ok.

probably-Erwins-Cat, 2023-12-28:
Hey HartmutA,
Unfortunately, there is a difference between the CAD models of various Raspberries and the physical one itself... Now, with the release of the RPI5, the same will repeat again. I modeled all parts after the CAD models, but they never fit the real-life part (same happens with Arducam).
I will update the RPI covers when I have all three relevant RPIs in my hand. I updated the covers provisionally, but I have the feeling that some others might now have fitting issues.

closed 2023-12-28 by probably-Erwins-Cat