OpenScan-org/OpenScan-Design #23: [BUG] An STL file for Classic has broken dimensions

FrenkK, 2024-02-19:
**Summary**

The file https://github.com/OpenScan-org/OpenScan-Design/blob/main/OS-Classic/V1/Pi%20Shield_RPi_with_Ringlight_and_Network%2BUSB.stl has a problem with dimensions, it is extremely large. I tried to convert from imperial, but it does not help. 

**Screenshots/Video**

[image]

**What behavior did you expect? vs. what happened?**

I expected the part to have somewhat similar dimensions to the smaller shield.

**Steps to reproduce**

Try to import the part in Prusaslicer.

**Log files**

N/A

Aucklandadrian, 2024-02-20:
Same issue

I just printed a number of parts and this one comes out about 10-15% too big
[image]

probably-Erwins-Cat, 2024-02-29:
The PR that properly rescales the part was merged. The described issue should now be solved

closed 2024-02-29 by probably-Erwins-Cat