OpenScan-org/OpenScan-Design #30: [BUG]Wrong hole geometry on 5-electronics-cover--green-shield.stl for both Mini and Midi

FelixTheCrazy, 2024-05-13:
**Summary**

The holes for the Raspberry Pi are not aligned properly with the 5-electronics-cover--green-shield.stl for both the Mini and Midi.

**Screenshots/Video**

Please add a screenshot and/or video that demonstrate the malfunction.
[image]

**What behavior did you expect? vs. what happened?**

Them to line up?

**Steps to reproduce**

Print it and try and fit the Pi there.

Assume that the developers know nothing: share the steps to recreate the bug.

**Log files**

probably-Erwins-Cat, 2024-06-26:
I will check the part. According to the CAD model of the RPi etc. It should align and we corrected the alignement somewhat recently.

probably-Erwins-Cat, 2024-06-26:
should be solved now. I uploaded the wrong part last time.
[image]

closed 2024-08-16 by probably-Erwins-Cat