OpenScan-org/OpenScan-Design #32: [DOKU] - Unclear which files to use

knicholes, 2024-05-18:
When trying to print the pi cam v2 cover, it's unclear which cover to use.  There is no stl for the camera/mount in https://github.com/OpenScan-org/OpenScan-Design/tree/main/OS-Mini/V2

There ARE STLs in
https://github.com/OpenScan-org/OpenScan-Design/tree/main/OS-Mini/V2/legacy-deprecated/RC1
(NOT RC2)
https://github.com/OpenScan-org/OpenScan-Design/tree/main/OS-Mini/V2/legacy-deprecated/beta1
https://github.com/OpenScan-org/OpenScan-Design/tree/main/OS-Mini/V2/legacy-deprecated/beta2

The dates for all of these files are all "2 months," so it's not clear to me which is newer.  I can guess that either beta2 or RC1 are the newest by the numbering, but it'd be great to know which has been used and works before printing it out. The fact that beta2 and RC1 are under "legacy-deprecated" makes me think I'm looking in the wrong place, though.

probably-Erwins-Cat, 2024-06-26:
"deprecated" means those parts are not in use anymore and shouldn't be printed for new builds. They were replaced by up-to-date parts in the base folder /V2/. They deprecated parts are left in this folder for the early adopters of the Mini/Midi V2 so that they do not have to reprint all parts, just because they broke the IU and the newer IU has slightly different dimensions. Those folders will eventually be removed when the last early adopters re-printed the up-to-date parts.

The Imaging Unit (IU) is compatible with the Mini and Midi V2. Whenever I find time designing a Maxi version, I also guarante that the imaging unit will be compatible with that design. Thus, I placed the imaging unit in a separate folder. 
Further, there might also be the option to design an alternative imaging unit, that holds your smartphone instead of a separate camera. As open-source project, we want to leave that modular design open for users.

In short, ignore the "deprecated" and "legacy" parts, they are for the early adopters, and jump directly to the new parts. The minimal building manual also has the same numbering scheme as the parts, e.g. **1**-IU-cover.stl which is the **1**st part in the building process of the IU.

probably-Erwins-Cat, 2024-08-16:
Updated the folder structure, so it is slightly less confusing

closed 2024-08-16 by probably-Erwins-Cat