OpenScan-org/OpenScan-Design #39: [BUG] Imaging Unit frame does not fit on rotor

Bob-vdV, 2025-03-27:
**Summary**
The current version of the imaging unit frame [file](https://github.com/OpenScan-org/OpenScan-Design/blob/main/files/Imaging-Unit/STL/3-IU-frame%20(Meshed).stl) STL does not fit on the rotor. The gap between the two holes is too small for the rotor to fit. The IU frame from the assembly instructions does not have one of the ends and just attaches to the side of the rotor. Could the IU file be reverted to this previous version?

**IU frame from assembly images**
[image]

**Latest IU frame**
[image]

Bob-vdV, 2025-03-28:
Just realized the 'slot' on the rotor is actually built in support, but it is very hard to remove and I had to cut it out. Could it be changed to something smaller, such as only the outer half of the cylinder? That would make it a bit more obvious it's support and also easier to remove, while it should still be enough to support the rest of the print.

probably-Erwins-Cat, 2025-03-30:
The same feedback was also provided from other users. The built-in support will be reworked to make it more fragile in a future update!

probably-Erwins-Cat, 2025-05-06:
should be fixed now. the built-in support is now clearly visible and easier to remove.

closed 2025-05-06 by probably-Erwins-Cat